chili-chips-ba/openeye-CamSI #1: HDMI

IsamVrce, 2024-05-03:
The main limiting factor in the HDMI design regarding resolutions and frame rates support is data bandwidth. We use [timing calculator](https://tomverbeure.github.io/video_timings_calculator) to calculate adequate pixel clock for desired frame rate. In [timing calculator](https://tomverbeure.github.io/video_timings_calculator) it is possible to chose between various standards: 
- CVT
- CVT-RB 
- CVT-RBv2 
- CEA-861 
- DMT
In the beginning we used CVT-RBv2, but after custom configuration is utilized

After choosing one of them all necessary parameters, for HDMI are calculates, among them is pixel clock.
> [!NOTE]
> Due to our RTL implementation pixel clock calculated here needs to be multiplied by 5

Because of this and the fact that PLL in our FPGA "Artix-7 csg324-2" can produce maximum of 628MHz, if the driver for particular driver is set to be BUFG, maximum pixel clock we can use is 125.6MHz. Pixel clock of 125.6MHz is not enough to reproduce output image of 1080p@60Hz, but it is enough for every resolution listed below:

- [ ] 1920 x 1080 30Hz
- [ ] 1280 x 720 120Hz
- [ ] 1280 x 720 60Hz
- [ ] 1024 x 768 60Hz
- [ ] 800 x 600 60Hz
- [ ] 640 x 480 60Hz

suarezvictor, 2024-05-27:
Hi, I usually do HDMI output in the Artix-7 at 1920x1080 60Hz (that's 148MHz pixel clock) using LiteX
do you want me to provide an example?

chili-chips-ba, 2024-05-28:
Yes, please provide an example, along with complete Vivado timing report (since the issue is not about making it work, but ensuring that it works reliably across PVT).

chili-chips-ba, 2024-06-06:
@suarezvictor any update on the example for Artix-7 1920x1080@60Hz HDMI (=**148MHz** pixel clock)  with complete Vivado timing report?!

suarezvictor, 2024-06-06:
Could you wait a bit more until I finish some pending, paid work I need to deliver? I need that to make a living, Then I'll provide the test as promised
by the way, even if PVT if not meet (but let me test that), the bit error rate will be so low that would pass unseen by the eye. Note that this is to be consumed by a person looking to a display, the application is not a network one.

I still recommend not limited to less than 1080p since it works in practice.

suarezvictor, 2024-06-06:
I haven't tested audio, and only used the DVI protocol. No audio, but simpler, accepted by all HDMI displays, and no licensing issues

AnttiLukats, 2024-06-13:
I am afraid if you use HDMI connector, HDMI cable and HDMI monitor then HDMI license fees apply no matter you use DVI protocol.

suarezvictor, 2024-06-13:
I don' share that idea. If I purchase an HDMI cable, I can do use it without extra payment. Same for a monitor and purchased connector. They can license the HDMI protocol and logo, but DVI doesn't require that.

suarezvictor, 2024-06-13:
Basic test code is [here](https://gist.github.com/suarezvictor/d87b7a4c59a3a4e536d20f8e06b5c13b)

chili-chips-ba, 2024-06-14:
Do you happen to have Vivado timing reports? LiteX is choking on that Python...

closed 2024-09-13 by chili-chips-ba

chili-chips-ba, 2024-11-28:
@suarezvictor, @Juninho99 took your advice and defied Xilinx datasheet - Here is **1920x1080@60FPS RGB888** streaming from Sony IMX283 camera SOC to HDMI!

[image]

What helped the most is that the `reach of the x5 pixel clock was limited to only 4 IOSERDES primitives`. 

Given such a low fanout, we even considered running it unbuffered. Unfortunately, due to a number of idiosyncrasies with Trenz hardware, the 4 HDMI IOSERDES were not collocated, but spread within the IO Bank. We therefore opted not to take chances with hold time closure, or its variations from build to build. We instead pushed the BUFG beyond its stated Fmax, but counted on the short distance and only 4 loads it needed to cover. While this may still not work over PVT, the insignificant fanout reduces the risk of failure.

suarezvictor, 2024-12-04:
I'm super glad you made progress for 1080p! And more if I was of help ;-)
I cannot recall if the route I took is the same as you. Will you share some more details about the solution, like some code?

Juninho99, 2024-12-04:
You can also check out these modules in SystemVerilog https://github.com/chili-chips-ba/openeye-CamSI/tree/main/1.hw/lib/ip/hdmi.