chili-chips-ba/openeye-CamSI #4: Pinout inconsistencies and Debayer module

Juninho99, 2024-05-04:
**Overview:**
While working with the TEB0707-02 FPGA Carrier Card and integrating the debayer module for raw to BGR conversion, our team encountered pinout inconsistencies across CRUVI sites and different board versions. These inconsistencies necessitate creating different builds for each board version and incorporating RTL inverters to accommodate the variations in Pin (P) and No-Connect (N) assignments. This issue aims to address the pinout inconsistencies, understand the implications of board version differences, and adapt the debayer module accordingly.

**Background:**
The debayer module is responsible for converting raw image data captured by the camera into BGR (Blue-Green-Red) format ([Debayer](https://documentation.euresys.com/Products/MultiCam/MultiCam/Content/MultiCam_6_7_HTML_Documentation/100281.htm)), following the primary color filter arrangement of the image sensor (page 89 [IMX219](https://github.com/chili-chips-ba/openeye-CamSI/issues/0.doc/Sensor.2-lane.RPi2.1/IMX219PQ.Datasheet.pdf)). The module utilizes line buffering to process raw lines and generate BGR pixels, which are then sent to the FIFO buffer for further processing and transmission.

**Pinout Inconsistencies:**
The CRUVI FPGA Carrier Card exhibits pinout inconsistencies across CRUVI sites and different board versions, posing challenges for hardware integration and RTL design. Despite efforts to manage these inconsistencies through RTL inverters, the variations in Pin (P) and No-Connect (N) assignments necessitate creating separate builds for each board version.

**Manufacturer Response:**
The manufacturer acknowledges the pinout inconsistencies and explains that inconsistencies in Pin (P) and No-Connect (N) assignments are inherent to the design of the 4x5 System on Modules (SoMs) and are mirrored in the baseboard. The flexibility in P/N mapping aims to optimize routing and accommodate different board configurations.

**Debayer Module Adaptation:**
In light of the pinout inconsistencies and board version differences, the debayer module requires adaptation to ensure compatibility across different CRUVI FPGA Carrier Card versions. Additionally, considerations for converting BGR format to other color spaces (e.g., RGB, HSV, XYZ) may be explored to enhance flexibility and compatibility with diverse applications.

**Next Steps:**
- Modify the debayer module to dynamically adjust to pinout variations and ensure compatibility with different board versions.
- Investigate options for implementing color space conversion modules (e.g., BGR to RGB, BGR to HSV) to enhance the versatility of the debayer module.
- Conduct thorough testing and validation to verify the functionality and compatibility of the adapted debayer module across different board versions.

chili-chips-ba, 2024-06-06:
Validate whether this issue in Trenz hardware is sufficiently addresses by **DINVERT** parameter in `top_pkg.sv`. Consider closing if so.

Juninho99, 2024-06-06:
After we validate on the hardware that everything is fine, then we will close this issue.

chili-chips-ba, 2024-09-13:
@Juninho99 given that this bug was fixed long time ago, please consider closing the ticket.

Juninho99, 2024-11-11:
Everything was tested on 2-lane and 4-lane camera sensor, and **DINVERT** did his job successfully.
Issue can be closed.

closed 2024-11-11 by Juninho99