chili-chips-ba/openeye-CamSI #10: Accumulated Cocotb issues/feature requests

rmetchev, 2024-05-16:


chili-chips-ba, 2024-05-17:
Xilinx sim models should be sanitized so that they don't polute log with irrelevant warnings, such as:
`%Warning-UNDRIVEN: ../../models/no-change/glbl.v:19:10: Signal is not driven: 'p_up_tmp'
                                                      : ... note: In instance 'top.glbl'
   19 |     tri1 p_up_tmp;
      |          ^~~~~~~~
                   ... For warning description see https://verilator.org/warn/UNDRIVEN?v=5.024
                   ... Use "/* verilator lint_off UNDRIVEN */" and lint_on around source to disable this message.
%Warning-UNUSEDSIGNAL: ../../models/no-change/glbl.v:20:26: Signal is not used: 'PLL_LOCKG'
                                                          : ... note: In instance 'top.glbl'
   20 |     tri (weak1, strong0) PLL_LOCKG = p_up_tmp;
      |                          ^~~~~~~~~
%Warning-BLKSEQ: ../../models/no-change/IDELAYCTRL.v:100:16: Blocking assignment '=' in sequential logic process
                                                           : ... Suggest using delayed assignment '<='
  100 |       attr_err = 1'b1;
      |                ^
%Warning-INITIALDLY: ../../models/no-change/IDELAYCTRL.v:124:15: Non-blocking assignment '<=' in initial/final block
                                                               : ... This will be executed as a blocking assignment '='!
  124 |    clock_edge <= 0;
      |               ^~
%Warning-INITIALDLY: ../../models/no-change/IDELAYCTRL.v:125:15: Non-blocking assignment '<=' in initial/final block
                                                               : ... This will be executed as a blocking assignment '='!
  125 |    clock_high <= 0;
      |               ^~
%Warning-INITIALDLY: ../../models/no-change/IDELAYCTRL.v:126:14: Non-blocking assignment '<=' in initial/final block
                                                               : ... This will be executed as a blocking assignment '='!
  126 |    clock_low <= 0;
      |              ^~
%Warning-INITIALDLY: ../../models/no-change/IDELAYCTRL.v:127:9: Non-blocking assignment '<=' in initial/final block
                                                              : ... This will be executed as a blocking assignment '='!
  127 |    lost <= 1;
      |         ^~
%Warning-INITIALDLY: ../../models/no-change/IDELAYCTRL.v:128:11: Non-blocking assignment '<=' in initial/final block
                                                               : ... This will be executed as a blocking assignment '='!
  128 |    period <= 0;
      |           ^~
%Warning-INITIALDLY: ../../models/no-change/IBUFDS.v:64:39: Non-blocking assignment '<=' in initial/final block
                                                          : ... This will be executed as a blocking assignment '='!
   64 |             "FALSE" : DQS_BIAS_BINARY <= #1 1'b0;
      |                                       ^~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.o_out_divide' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.count' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.first_rise' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.half_period_done' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.ce_enable1' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.ce_enable2' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.ce_enable3' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
  128 |     always @(gsr_in or clr_in)
      |     ^~~~~~
%Warning-LATCH: ../../models/BUFR.v:128:5: Latch inferred for signal 'top.u_csi_rx_top.u_phy_clk.u_bufr.ce_enable4' (not all control paths of combinational always assign a value)
                                         : ... Suggest use of always_latch for intentional latches
[... cut ...]
      |        ^~~~
%Warning-MULTIDRIVEN: ../../models/no-change/IDELAYCTRL.v:84:24: Signal has multiple driving blocks with different clocking: 'top.u_clkrst_gen.u_idelayctrl.clock_negedge'
                      ../../models/no-change/IDELAYCTRL.v:168:4: ... Location of first driving block
  168 |    clock_negedge <= 1'b0;
      |    ^~~~~~~~~~~~~
                      ../../models/no-change/IDELAYCTRL.v:156:4: ... Location of other driving block
  156 |    clock_negedge <= 1'b1;
      |    ^~~~~~~~~~~~~`

chili-chips-ba, 2024-05-17:
_'glbl.v'_ and other Xilinx models should be copied into our project filesystem (e.g. `_2.sim/models_`) and used from there. We cannot assume or expect that everyone would have Xilinx tools installed in the same path, yet alone use the same Vivado version as the one that's hard-coded in our _Makefile_ 
[image]

chili-chips-ba, 2024-05-17:
[image]

What exact sim tool (including version) failed to catch this basic syntax error?

chili-chips-ba, 2024-05-17:
make SIM=verilator WAVES=1
[image]

[image]

rmetchev, 2024-05-17:
> ![image](https://private-user-images.githubusercontent.com/67533663/331479433-df939c40-9341-4456-96dc-bc98c90628a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Fixed.
Verilator let that go through.

rmetchev, 2024-05-17:
> Xilinx sim models should be sanitized so that they don't polute log with irrelevant warnings, such as: `%Warning-UNDRIVEN: ../../models/no-change/glbl.v:19:10: Signal is not driven: 'p_up_tmp' : ... [...]

There are ways to filter out warnings, "sanitizing" vendor IP, esp if not absolutely necessary (ie fatal error) is not one of them, and is a recipe for disaster,

chili-chips-ba, 2024-07-19:
This activity is now transferred to [Issue19](https://github.com/chili-chips-ba/openeye-CamSI/issues/19)

closed 2024-09-13 by chili-chips-ba