chili-chips-ba/openeye-CamSI #19: issue#5

Juninho99, 2024-06-08:
# Issue: Develop Testbench Environment for Testing CSI-2 Camera Model in Cocotb

## Description

The goal of this issue is to develop a testbench environment for testing a CSI-2 camera model using Cocotb. The primary objective is to create a Python class that simulates the actual operation of a camera by generating and sending `cam_dphy_dat` and `cam_dphy_clk` signals. These signals are differential inputs to the top module, and the class will be responsible for setting these signals to specific values.

## Tasks

### 1. Basic Signal Generation
- [ ] **Create a basic function to generate random signals for `cam_dphy_dat` and `cam_dphy_clk`**
  - This function should produce simple signal patterns to observe their behavior in the simulation.
  - Verify the correct behavior and timing of these signals in the simulation environment.

### 2. Camera Simulation Class
- [ ] **Design and implement a Python class to simulate camera behavior**
  - The class should handle the generation of differential signals for `cam_dphy_dat` and `cam_dphy_clk`.
  - Implement methods to set these signals to desired values for different test scenarios.

### 3. Data Transmission Functions
- [ ] **Implement functions to send specific bits and bytes**
  - Create methods in the camera simulation class to send individual bits and bytes over the differential signals.
  - Ensure these functions respect the number of data lanes specified as a parameter.

### 4. Line and Frame Control Functions
- [ ] **Develop functions for sending start/end of line and frame signals**
  - Add methods to handle the transmission of signals indicating the start and end of lines and frames.
  - Implement mechanisms for sending line and frame data according to the CSI-2 protocol.

### 5. Integration and Testing
- [ ] **Integrate the class and functions into the testbench environment**
  - Ensure that the class can be easily instantiated and used within Cocotb tests.
  - Write initial tests to validate the functionality of the class and its methods.

### 6. Documentation
- [ ] **Document the class and its usage**
  - Provide comprehensive documentation on how to use the class and its methods.
  - Include example test cases and expected outputs.

## Expected Outcome

By completing this issue, we expect to have a fully functional testbench environment capable of simulating the CSI-2 camera model's behavior. This environment will facilitate thorough testing and validation of the top module's interaction with the camera signals, ensuring robust and reliable operation.

chili-chips-ba, 2024-06-08:
In addition to generating random data patterns, add an option for **Incrementing Data Pattern**. Such option would checking the downstream data integrity easier. It would also facilitate the process of following image transformations when looking at the waves.

Also add a new task category: **Generate Errors and Corner Cases**. For the starters, within it, add options for inserting:
- lane-to-clock skew
- lane-to-lane skew
- clock jitter

These options can be specified as the max deviation from the nominal, and the model would then randomly deviate within that range.

Within _Line and Frame Control Functions_ category, provide an option to specify _Resolution and FPS_. For the starters, this set needs to include at least the rates supported by our current RTL:
- `define **HDMI_720p60**
- `define **HDMI_1080p30**
- `define **HDMI_1080p60** , see [Issue#1](https://github.com/chili-chips-ba/openeye-CamSI/issues/1) and @suarezvictor comment

Juninho99, 2024-06-13:
On current branch [**issue#5**](https://github.com/chili-chips-ba/openeye-CamSI/tree/issue%235) we have implemented basic functions for control signals `cam_dphy_dat` and `cam_dphy_clk`. We can send a sequence of bytes on the data lane, and everything is received properly through RTL. For example, through cocotb we send this sequence:

```python
   sequence = [
      ([0b00000000, 0b11111111], 40), # Some random data at the start
      ([0b10111000, 0b01000111], 20), # Sync bytes b8b8
      ([0b00010010, 0b11101101], 20)  # Start of long packet
   ]
```    

After the simulation is done, we can obtain results and see that everything works fine, except for the part of the `packet_handler` where RTL didn't set `long_packet` signal in right way. In the red block, we can see that all bytes have been received properly.

[image]

From folder `/2.sim/cocotb/top` you can run the simulation with the command: `make SIM=icarus WAVES=1 TESTCASE=test_1`
Also from the same path you can open gtkwave with: `gtkwave sim_build/top.fst` and inside it, you can read `csi-2.gtkw` from the `/2.sim/cocotb/top folder.

chili-chips-ba, 2024-06-14:
**__1__** 
For readability, consider using Hex instead of binary format in stimulus. That would not only shorten the line, but also allow removing b8b8 from the comment:
`([0b10111000, 0b01000111], 20), # Sync bytes b8b8`

**__2__** 
- What's the plan for scaling the stimulus from 2 lanes to 4 lanes? 
- Can it be made scalable from the get-go from one place, by one-and-only-one control knob?!

Juninho99, 2024-06-25:
First of all, the simulation is now working pretty much good. Everything was done using Verilator. Of course, there were many problems regarding the choice of the simulator, but from this point CSI model can be continued.

So, as we can see from the image below, `csi_in_frame` and `csi_in_line` signals are there and they behave as expected. All sync bytes are cought properly, and all data from python script are received.

[image]

Current functions in python script are not so edited nor readable. However, the first task from this issue is done and tested successfully. Next tasks are related to the class and much prettier way of sending data.

For curiosity, here are some lines of code that send sequence of data to the differential signal `cam_dphy_dat`:

```python
   sequence = [
      ([0x00, 0xFF], 2),    # Some random data at the start
      ([0xB8, 0x47], 1),    # Sync bytes b8b8
      ([0x12, 0xFF], 1),    # Start of frame (FF inv. -> 00)
      ([0x81, 0x7E], 3),    # Random data
      ([0xFF, 0x00], 2),    # Random data
      ([0x00, 0xFF], 2),    # Random data
      ([0xB8, 0x47], 1),    # Sync bytes b8b8
      ([0x32, 0xED], 1),    # Start of long packet (32h bytes, ED inv. -> 12) - EMBEDDED DATA
      ([0x11, 0xFF], 1),    # (FF inv. -> 00), combined with previous 32 --> 0032 bytes to read
      ([0xFF, 0x00], 32),   # Data for read
      ([0x00, 0xFF], 2),    # Random data
      ([0xB8, 0x47], 1),    # Sync bytes b8b8
      ([0x32, 0xD7], 1),    # Start of long packet (32h bytes, D7 inv. -> 28) - START OF LINE
      ([0x11, 0xFF], 1),    # (FF inv. -> 00), combined with previous 32 --> 0032 bytes to read
      ([0x22, 0xDD], 32),   # Data for read
   ]
```    

Definitely, this can be done more efficiently and that will be done in the next tasks.

About your previous suggestions @chili-chips-ba:
1. Surely more readable with 0x..... and comments will be removed later
2. We will get to it. With class and methods, I think it won't be a problem
3. More luck with Verilator, but also with some problems, which will be discussed in next issues
4. Yes. It is already there, we can make it with `make g`, but definitely I'll make `Makefile` more clearly
5. Stay tuned 😉

Juninho99, 2024-06-25:
I'll mention some of issues I encountered with the simulators.

**Icarus Verilog Issues:**
Development of the testbench with Icarus has stalled since our current RTL supports more advanced SystemVerilog features that Icarus cannot support. I am curious if there will be any progress in this area soon, particularly regarding indexing, for loops, etc.

Additionally, regarding the csi_rx_align_word.sv module, the signals `word_in_pipe` and `valid_in_pipe` do not appear at all in the simulation. Consequently, the rest of the simulation with Icarus could not function properly. The reason for this is the use of "packed" arrays for these signals, while "unpacked" arrays do display the signals.

**Verilator Issues:**
Several issues also emerged with Verilator. The first issue is related to the fact that some XILINX primitives can't be simulated with it. Specifically, the IDELAYE2 component is currently ignored, and it will need to be written separately later to function as expected.

Moreover, there was a problematic part of the simulation from cocotb concerning the differential signal cam_dphy_dat. I could not assign its value intuitively as I did using Icarus as the simulator:

```python
signal[lane*2 + 1].value = bit_value  # Positive bit
signal[lane*2].value = ~bit_value & 0x1  # Negative bit (inverse)
```    

With Icarus, I created two separate coroutines to set the signal values for each line in parallel. This approach did not work with Verilator. The only way to set the signal values was to assign values to all bits in one coroutine, like this:

```python
bit_a = (byte0 >> bit) & 0x1
bit_b = ~bit_a & 0x1
bit_c = (byte1 >> bit) & 0x1
bit_d = ~bit_c & 0x1

# Set the signal values for both lanes
signal.value = (bit_b << 3) | (bit_a << 2) | (bit_d << 1) | bit_c
```    
Additionally, below is the image showing the specific error when I tried to use the first code snippet with the Verilator.
[image]

I hope I have "overlooked" something in these mentioned points and that a solution exists or that there is active work on these problems.

I think there have been discussions on similar topics here:

[Verilator Pull Request #2728](https://github.com/verilator/verilator/pull/2728)
[Cocotb Issue #2380](https://github.com/cocotb/cocotb/issues/2380)
[Verilator Issue #2727](https://github.com/verilator/verilator/issues/2727)

Juninho99, 2024-06-29:
**Update: Implementation of CSI Class for Camera Simulation**

We have implemented a `CSI` class as a foundational component for future camera functionality testing. This class has been thoroughly tested using Verilator. It includes methods such as `send_combined_bytes`, `send_data_pattern`, `start_frame`, `end_frame`, `send_embedded_data`, `send_line`, `send_frame`, and `run`. During testing, the class was instantiated with parameters such as (you can find this at the bottom of `test_2` in `/2.sim/cocotb/top/test_top.py`):

```python
num_lane = 2
dinvert = [0, 1] if num_lane == 2 else [0, 1, 0, 1]
    
csi = CSI(dut, dut.cam_dphy_dat, period_ns=1.096, fps_Hz=60, line_length=1280, frame_length=720, frame_blank=130, num_lane=num_lane, dinvert=dinvert)

await csi.run(3)  # Sending 3 frames as an example
```    

Extensive testing has been performed, and attached is an image showing three frames sent, each containing 5 lines (a reduced number for testing purposes).
[image]

The functionality has been predominantly validated for 2 lanes, with partial implementation and testing for 4 lanes pending until the RTL is tested/corrected for 4 lanes. Additionally, the `dinvert` feature has been tested and behaves as expected.

Further detailed testing and documentation are required. Currently, tests can be run in the `/2.sim/cocotb/top` folder using the command `make WAVES=1 TESTCASE=test_2`. After the test completes, open GTKWave with `gtkwave dump.fst` in the same folder. The `csi-2-verilator.gtkw` file in GTKWave provides a comprehensive view of all necessary signals.

All of those additional codes are in the issue#5 branch.

Juninho99, 2024-06-30:
**ECC successfully tested**

Module csi_rx_packet_handler reverted to original state with ecc part successfully tested. In the CSI class, a part for calculating ECC has been added, but which should be adapted to the specifics of the camera and the data being sent. On the test example, everything works as expected. Here is a function to calculate ecc in python:

```python
def calculate_ecc(self, byte1, byte2, byte3):
      """
      Calculate the ECC for the given three bytes.
      The ECC is computed over the 24 bits formed by concatenating the three bytes.
      """
      data = (byte1 << 16) | (byte2 << 8) | byte3

      ecc = 0

      ecc_bit5 = ((data >> 10) & 1) ^ ((data >> 11) & 1) ^ ((data >> 12) & 1) ^ ((data >> 13) & 1) ^ ((data >> 14) & 1) ^ ((data >> 15) & 1) ^ ((data >> 16) & 1) ^ ((data >> 17) & 1) ^ ((data >> 18) & 1) ^ ((data >> 19) & 1) ^ ((data >> 21) & 1) ^ ((data >> 22) & 1) ^ ((data >> 23) & 1)
      ecc |= ecc_bit5 << 5

      ecc_bit4 = ((data >> 4) & 1) ^ ((data >> 5) & 1) ^ ((data >> 6) & 1) ^ ((data >> 7) & 1) ^ ((data >> 8) & 1) ^ ((data >> 9) & 1) ^ ((data >> 16) & 1) ^ ((data >> 17) & 1) ^ ((data >> 18) & 1) ^ ((data >> 19) & 1) ^ ((data >> 20) & 1) ^ ((data >> 22) & 1) ^ ((data >> 23) & 1)
      ecc |= ecc_bit4 << 4

      ecc_bit3 = ((data >> 1) & 1) ^ ((data >> 2) & 1) ^ ((data >> 3) & 1) ^ ((data >> 7) & 1) ^ ((data >> 8) & 1) ^ ((data >> 9) & 1) ^ ((data >> 13) & 1) ^ ((data >> 14) & 1) ^ ((data >> 15) & 1) ^ ((data >> 19) & 1) ^ ((data >> 20) & 1) ^ ((data >> 21) & 1) ^ ((data >> 23) & 1)
      ecc |= ecc_bit3 << 3

      ecc_bit2 = ((data >> 0) & 1) ^ ((data >> 2) & 1) ^ ((data >> 3) & 1) ^ ((data >> 5) & 1) ^ ((data >> 6) & 1) ^ ((data >> 9) & 1) ^ ((data >> 11) & 1) ^ ((data >> 12) & 1) ^ ((data >> 15) & 1) ^ ((data >> 18) & 1) ^ ((data >> 20) & 1) ^ ((data >> 21) & 1) ^ ((data >> 22) & 1)
      ecc |= ecc_bit2 << 2

      ecc_bit1 = ((data >> 0) & 1) ^ ((data >> 1) & 1) ^ ((data >> 3) & 1) ^ ((data >> 4) & 1) ^ ((data >> 6) & 1) ^ ((data >> 8) & 1) ^ ((data >> 10) & 1) ^ ((data >> 12) & 1) ^ ((data >> 14) & 1) ^ ((data >> 17) & 1) ^ ((data >> 20) & 1) ^ ((data >> 21) & 1) ^ ((data >> 22) & 1) ^ ((data >> 23) & 1)
      ecc |= ecc_bit1 << 1

      ecc_bit0 = ((data >> 0) & 1) ^ ((data >> 1) & 1) ^ ((data >> 2) & 1) ^ ((data >> 4) & 1) ^ ((data >> 5) & 1) ^ ((data >> 7) & 1) ^ ((data >> 10) & 1) ^ ((data >> 11) & 1) ^ ((data >> 13) & 1) ^ ((data >> 16) & 1) ^ ((data >> 20) & 1) ^ ((data >> 21) & 1) ^ ((data >> 22) & 1) ^ ((data >> 23) & 1)
      ecc |= ecc_bit0

      return ecc
```

chili-chips-ba, 2024-07-02:
@Juninho99 great work on creating:
- the first-ever opensource model for the CSI2 camera
- simulation framework and bring up

The next step is to distill some of this info into a brief, yet informative README.md section on how to run the simulation and modify it to per user needs, with a few examples, as shown above.

Please, also bring up the issues you've found with Verilator, cocotb and Icarus to their relevant owners.

chili-chips-ba, 2024-07-09:
Let's also make the tests _auto-checking_, so that the designer does not need to eye-ball the waveforms for every sim run, but instead the test itself finds and reports issues in RTL

Juninho99, 2024-07-09:
Yes, definitely we need to provide some tests that will be auto-checked. It will be much easier for users to see where the issue is. Also we will briefly explain everything according to our model and scripts, but first of all we want to make sure that all the tasks with this issue are done.

closed 2025-04-24 by chili-chips-ba