chili-chips-ba/openeye-CamSI #21: Review of csi_rx_align_word.sv module

Juninho99, 2024-06-30:
We need to review the implementation of `csi_rx_align_word.sv` module. In corrected version, that I proposed in the [issue#5](https://github.com/chili-chips-ba/openeye-CamSI/blob/issue%235/1.hw/csi_rx/csi_rx_align_word.sv) branch, everything works properly in the simulation. However, when I use the version that is currently on the [main](https://github.com/chili-chips-ba/openeye-CamSI/blob/main/1.hw/csi_rx/csi_rx_align_word.sv) branch, something isn't right. Below are images that show the difference. 

In the first image, we can see that `word_out` is delayed with two clock cycles, which is preferably.
[image]

In the second image, there is only one clock cycle delay, which causes with subsequent failures.
[image]

Also, other signals don't behave in the right direction (`wait_for_sync`, `packet_done`, etc.)

chili-chips-ba, 2024-07-02:
`valid_out` in the first (aka "good wave") is not covering the B8. Is that expected?

chili-chips-ba, 2024-07-19:
The original bug was about misalignment of VLD and DATA, and simple to fix. However we still cannot declare the _align_word_ module fully tested, as the existing testbench does not exercise its primary function. 

To close this issue, static lane-to-lane skew needs to be added to the traffic generator. The test should cover at least the following two cases:
1) when lane-to-lane skew is within `deskewing range of this module, which is 2 clock cycles`
2) when lane-to-lane skew exceed that range

Similarly, we cannot declare that _align_byte_ module is fully tested without introducing bit skew into traffic generator for each lane. 

@Juninho99, let's improve the sim setup with these must-have capabilities for further progress on this ticket.

chili-chips-ba, 2024-09-13:
@Juninho99 please consider closing this ticket. If there are any outstanding DV-related activities, open a new `task` issue to track them.

closed 2025-04-24 by chili-chips-ba