chili-chips-ba/openeye-CamSI #27: issue#8 - isp_top.sv cleanup

chili-chips-ba, 2024-09-13:
1) `RGB_WIDTH` is currently only partially parameterized. It includes the mix of internal hard-coded, and externally-settable values, which is the worst method of both worlds. We should either:
- eliminate the parameter, and use hard-coded values throughout
- or fully support the parameter

2) The currently used `ifdef` style for the selection of RAW format (8-bit, 10-bit, 12-bit) is sub-optimal. With clean and readable RTL as our objective, that little quirk should be corrected

Juninho99, 2024-11-30:
Maybe everything is okay now? In the future, only one module will be for raw2rgb

chili-chips-ba, 2024-11-30:
Closing this issue as the original problem is now corrected.

closed 2024-11-30 by chili-chips-ba