chili-chips-ba/openeye-CamSI #29: issue#9 - Implementing Variable Pauses After I2C Register Writes in i2c_ctrl Module

Juninho99, 2024-10-29:
## Description:
According to the camera's specification (**Sony IMX294**), after certain registers are written, there needs to be a pause of a few milliseconds before proceeding to the next set of registers.

## Objective:
Modify the existing `i2c_ctrl` module and `i2c_top` module to implement variable pauses after specific I2C register writes, as specified by the camera's requirements.

## Problem Statement:
The challenge is to introduce variable pauses after writing certain registers, as required by the camera's specification. These pauses are not periodic and occur only after specific registers. The pauses need to be controlled dynamically, allowing for different durations specified in milliseconds.

## Proposed Solution:
- Modify the `i2c_init.mem` file to include the pause duration for each register write.
- Update the `i2c_data_init` array in the `i2c_top` module to read the pause durations along with register addresses and data.
- Adjust the `i2c_ctrl` module to implement a **WAIT** state that counts the required pause duration using nested counters, avoiding resource-intensive multiplication.
- Ensure that pauses are dynamically controlled based on the data read from i2c_init.mem.

chili-chips-ba, 2024-10-30:
Differences between 2-lane and 4-lane cameras should not be hard coded in the main RTL body. Instead, they should go into customization package under corresponding branches, based on the single master selector.

[image]
-----
[image]

Juninho99, 2024-11-11:
You are right. I will make some changes and push them on issue#9 branch. Also I'll make pull request asap.

Juninho99, 2024-11-12:
The I2C part is fully parameterized and successfully tested on 4-lane sensor. However, we will keep this issue open until we have tested the I2C on both 2-lane sensors.

Juninho99, 2024-11-25:
IMX219 is also tested successfully. There is still a webcam left.

chili-chips-ba, 2025-04-24:
Hi @Juninho99 what is the latest status on this track? Can we now close it?

closed 2025-04-24 by Juninho99