chili-chips-ba/openeye-CamSI #31: openXC7 - slave_oserdese has disconnected OQ/OFB output ports

kerimbavcic, 2024-11-05:
> NOTE: when using Vivado, the issue is not present!

When using openXC7 (place and route step) we get an error in **OSERDESE2** module, **slave_oserdese** to be specific, where it says OQ and OFB ports are not connected. The exact message we get is:

```
ERROR: OSERDESE2 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.slave_oserdese' has disconnected OQ/OFB output ports
2 warnings, 1 error
make: *** [Makefile:53: top.fasm] Error 25
```

Here is the code snippet:
[image]

openXC7 installed version (https://github.com/openXC7/toolchain-installer): 
- commit hash - 53b1286
- yosys version - 0.38
- nextpnr version - 0.8.2

We noticed that openXC7 had some issues with OSERDESE2 before in the past which were fixed as part of this [bugfix](https://github.com/openXC7/openXC7-snap/releases/tag/0.7.0).

chili-chips-ba, 2024-11-06:
+ @hansfbaier, @MikeReznikov for comment

hansfbaier, 2024-11-06:
@chili-chips-ba Can I have a test design, that highlights your issue?

chili-chips-ba, 2024-11-06:
... RTL is already checked in. @kerimbavcic and @harunkovacevic are in the process of checking in the openXC7 part of the setup:
- https://github.com/chili-chips-ba/openeye-CamSI/tree/main/3.build/openXC7

They'll ping you shortly...

harunkovacevic, 2024-11-06:
@hansfbaier You can find all necessary files here: 

- https://github.com/chili-chips-ba/openeye-CamSI/tree/main/3.build/openXC7

kerimbavcic, 2024-11-06:
@hansfbaier We commited our source files and Makefile as well. You can find them at: "3.build/openXC7". Please have a look and let us know what you think :)

The problem is in "3.build/openXC7/1.hw/lib/fpgatech/XILINX/fpga_oser10.v".

Adam960926, 2024-11-07:
i have fix this error,  I'll submit the code in a moment

AdamLee7, 2024-11-07:
https://github.com/openXC7/nextpnr-xilinx/pull/53
I fixed the bug in the pack code, but due to other missing logic in packing and routing, and incomplete prjxray-db information, your netlist may still not work properly.

chili-chips-ba, 2024-11-07:
@AdamLee7 what is your recommendation for our next step? 
- Is it to simply pull in and try your patch
- then keep reporting issues as we come across them?!

hansfbaier, 2024-11-07:
@chili-chips-ba If your tests succeeds I will merge. The patch looks good to me so far.

AdamLee7, 2024-11-07:
> @AdamLee7 what is your recommendation for our next step?
> 
> * Is it to simply pull in and try your patch
> * then keep reporting issues as we come across them?!

yes, just do it, if  there is any problem, you can commit an issue

harunkovacevic, 2024-11-07:
We have cloned nextpnr-xilinx repo, adapted _xilinx/pack_io_xc7.cc_ file, compiled and installed a new nextpnr version. Then we ran a build and "disconnected ports OSERDESE2 issue" was fixed but we've gotten a new error explained in this [issue](https://github.com/chili-chips-ba/openeye-CamSI/issues/32).