chili-chips-ba/openeye-CamSI #32: openXC7 - slave_oserdese has illegal fanout on SHIFTIN1 or SHIFTOUT2

harunkovacevic, 2024-11-07:
When using openXC7 (place and route step) we get an error in OSERDESE2 module, slave_oserdese to be specific. 
The exact message we get is:
```
ERROR: OSERDESE2 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.slave_oserdese' has illegal fanout on SHIFTIN1 or SHIFTOUT2
2 warnings, 1 error
make: *** [Makefile:53: top.fasm] Error 255
```

These changes are included - [fix oserdes pack bug](https://github.com/openXC7/nextpnr-xilinx/pull/53) and it fixed previous [issue](https://github.com/chili-chips-ba/openeye-CamSI/issues/31).

harunkovacevic, 2024-11-07:
@hansfbaier, @AdamLee7 What could be the next steps?

hansfbaier, 2024-11-08:
The error message needs to be better there too. I can look into this on Monday

hansfbaier, 2024-11-10:
@harunkovacevic I added some cleanup and checks to @AdamLee7's contribution,
please use this new development branch:
https://github.com/openXC7/nextpnr-xilinx/commits/oserdes-master-slave/

harunkovacevic, 2024-11-11:
@hansfbaier After updating _pack_io_xc7.cc,_ now we have another error: 
```
ERROR: OSERDESE2 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.master_oserdese' can only be connected to the SHIFOUT port of another OSERDESE2, but is connected to port: 'SHIFTOUT1'
2 warnings, 1 error
make: *** [Makefile:53: top.fasm] Error 255
```

kerimbavcic, 2024-11-18:
@hansfbaier, @AdamLee7 Any updates guys? (maybe you missed Haruns last comment)

hansfbaier, 2024-11-18:
I already fixed that in my branch, but now the problem is that nextpnr-xilinx does not support BUFM

hansfbaier, 2024-11-18:
So you would need to test the branch and close this ticket if successful

chili-chips-ba, 2024-11-19:
@hansfbaier the branch should be also merged for the proper full closure of this issue.

hansfbaier, 2024-11-20:
@chili-chips-ba Branch is merged and changes are already live in the nix toolchain.