chili-chips-ba/openeye-CamSI #33: Vivado project does not seem to be up to date

hansfbaier, 2024-11-11:
When building with Vivado 2024.1 
it cannot synthesize correctly, because missing files/modules are reported.
It looks like those have not been added to the project yet.
Is the project up to date with the current sources?

Juninho99, 2024-11-12:
You can now pull the latest changes from the main branch and try building again. Let us know if you encounter any further issues.

chili-chips-ba, 2024-11-22:
@hansfbaier please close this issue if the original problem has now been resolved.

hansfbaier, 2024-11-22:
I get this weird error with Vivado 2024.1:
```
[Place 30-484] The packing of LUTRAM/SRL instances into capable slices could not be obeyed. 

	Number of LUTRAMs/SRLs: 3072
	Theoretically, assuming any two LUTRAMs/SRLs can be packed into a slice, the number of capable slices required is 768 out of 4750 in the device (utilization 16.1684%)
	Actual number of capable slices required is 7680 out of 4750 in the device (utilization 161.684%).

As a result, 20 or more LUTRAMs/SRLs failed to place.
Names of the first 20 LUTRAMs/SRLs:
u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_27_29 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_36_38 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_39_41 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_3_5 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_45_47 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_12_14 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_15_17 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_24_26 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_27_29 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_36_38 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_39_41 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_3_5 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_15_17 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_24_26 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_27_29 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_39_41 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_12_14 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_18_20 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_24_26 type RAM64M
u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_27_29 type RAM64M

The mentioned LUTRAMs/SRLs are constrained as below: (listing maximum of 20 LUTRAMs/SRLs per constraint)
An internal area constraint:
  u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_27_29
  u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_36_38
  u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_39_41
  u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_3_5
  u_isp/u_raw2rgb/line2_mem_reg_r1_320_383_45_47
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_12_14
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_15_17
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_24_26
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_27_29
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_36_38
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_39_41
  u_isp/u_raw2rgb/line2_mem_reg_r1_384_447_3_5
  u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_15_17
  u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_24_26
  u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_27_29
  u_isp/u_raw2rgb/line2_mem_reg_r1_448_511_39_41
  u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_12_14
  u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_18_20
  u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_24_26
  u_isp/u_raw2rgb/line2_mem_reg_r1_64_127_27_29

  Number of capable slices required* by this constraint: 7680
  Number of capable slices available in this constraint region: 650
  Utilization = 1181%

* - The number of capable slices required is computed under assumption that the LUTRAMs/SRLs can be perfectly packed in the constrained region. Also, each LUTRAM/SRL may be a macro containing multiple LUTRAM/SRL instances.

Resolution: Please analyze your design to determine if the number of LUTRAMs/SRLs can be reduced by combining multiple LUTRAMs into Block RAMs for example.

[Place 30-99] Placer failed with error: 'Could not place all lutrams'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.
```

chili-chips-ba, 2024-11-22:
@Juninho99 please don't let fixes for issues like this one (which you've already fixed) linger for too long on your dev branch.

Juninho99, 2024-11-30:
After everything was okay with 4-lane sensor, you can now try to build it again on Vivado. I hope everything will be fine.

chili-chips-ba, 2024-11-30:
@hansfbaier please pull the checked-in design files and try again. If the original problem is now fixed, close this issue.

hansfbaier, 2024-12-01:
Now it works, thanks!

closed 2024-12-01 by hansfbaier