chili-chips-ba/openeye-CamSI #37: openXC7 - no Bels remaining of type 'BUFR'

harunkovacevic, 2024-12-01:
When using openXC7 (place and route step) we get an error in BUFR module, 'u_csi_rx_top.u_phy_clk.u_bufr' to be specific. 

The message we received is:
`ERROR: Unable to place cell 'u_csi_rx_top.u_phy_clk.u_bufr', no Bels remaining of type 'BUFR' 2 warnings, 1 error make: *** [Makefile:53: top.fasm] Error 255`

This error arises from attempting to resolve this [issue.](https://github.com/chili-chips-ba/openeye-CamSI/issues/36) We have now replaced BUFMR with BUFR, and the design was successfully tested in Vivado. @hansfbaier, @MikeReznikov Are there any indications on how we can put an end to this problem?

chili-chips-ba, 2024-12-01:
Even if [BUFMR](https://github.com/chili-chips-ba/openeye-CamSI/issues/36) and BUFR are internally the same component, they externally have different features, different look-and-feel, and are also treated separately in the corresponding vendor documentation. 

We therefore want to track them as two separate, distinct issues.

kerimbavcic, 2024-12-02:
@AdamLee7 send us the PR link or the branch link when you finish please :)

kerimbavcic, 2024-12-08:
@AdamLee7 are there any news?

hansfbaier, 2024-12-08:
@kerimbavcic The primitives-fixes branch has the BUFR support by @AdamLee7 already integrated.
But we need to test this branch more before merging, especially the LUTRAM changes there.

hansfbaier, 2024-12-08:
@kerimbavcic The primitives-fixes branch has the BUFR support by @AdamLee7 already integrated, if I am not mistaken.
But we need to test this branch more before merging, especially the LUTRAM changes there.

chili-chips-ba, 2025-04-03:
@hansfbaier @regymm @edmundhumenberger how is this merge going?

Since we have worked out RTL for a clocking scheme that does not use BUFR, thus bypassing this openXC7 issue, if you provide a link you use to track it, we could close our issue. Does this sound good?

closed 2025-04-24 by chili-chips-ba