chili-chips-ba/openeye-CamSI #38: openXC7 - ERROR: Unrouteable $PACKER_GND_NET sink

harunkovacevic, 2025-02-08:
When building the project, the following error occurs:
`Routing arc 218 of net '$PACKER_GND_NET' (0, 0) -> (4, 94)
ERROR: Unrouteable $PACKER_GND_NET sink u_csi_rx_top.lane[0].u_phy_dat.ideser.D (SITEWIRE/ILOGIC_X0Y110/DINV_OUT)
2 warnings, 1 error
make: *** [Makefile:56: error1.fasm] Error 255`

Steps to reproduce:
- Navigate to _3.build/openXC7_, where the Makefile is located.
- Run `make error1`

@hansfbaier, @MikeReznikov Do you have any ideas on how to solve this problem?

chili-chips-ba, 2025-02-15:
@regymm , any additional tips or insights?

regymm, 2025-02-18:
I'm not able to reproduce this yet, I got 
```
Info: Preparing clocking...
Info: Packing constants..
ERROR: OSERDESE2 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.slave_oserdese' has disconnected OQ/OFB output ports)
```
instead, probably before the routing error you mentioned. 

But, the D port on ISERDESE2 is an input https://docs.amd.com/r/2021.1-English/ug953-vivado-7series-libraries/ISERDESE2
in the code it's tied to zero, you can try leaving it unconnected instead. 
In UberDDR3, the instantiation is like this: https://github.com/AngeloJacobo/UberDDR3/blob/main/rtl/ddr3_phy.v#L705

harunkovacevic, 2025-02-19:
When I run the same command again **without changing** anything, I sometimes get a different error:
```Rerouting arc 162 of net 'csi_byte_clk' without bounding box, possible tricky routing...  
Routing arc 162 of net 'csi_byte_clk' (33, 93) -> (78, 103)  
ERROR: Failed to route arc 162 of net 'csi_byte_clk', from SITEWIRE/BUFGCTRL_X0Y31/O to SITEWIRE/SLICE_X14Y110/CFFMUX_OUT.  
2 warnings, 1 error  
make: *** [Makefile:57: error1.fasm] Error 255 
```

I assume that _nextpnr-xilinx_ does not use a fully deterministic routing approach, and that each run results in slight variations in routing decisions. It seems like it's attempting different heuristics, and depending on the routing congestion, one issue manifests over another.

Another thing I noticed is that both errors involve critical signals—one related to GND (`$PACKER_GND_NE`T) and another related to clock routing (`csi_byte_clk`).

Would using a fixed seed, like` --seed X`, help in making the results deterministic? For example:
`nextpnr-xilinx --seed 1 --chipdb ${CHIPDB}/${PART}.bin --xdc $(XDC_FILE) --json error1.json --fasm error1.fasm --verbose --debug
`

Also, would disabling `-abc9` in synth_xilinx during synthesis make a difference?

As a side note, unrelated to the main issue, I’m wondering if non-deterministic errors in FPGA design are often a sign of resource constraints or routing algorithm limitations. Instead of enforcing fully deterministic behavior, it might be more useful to analyze these variations to identify underlying problems.

hansfbaier, 2025-02-24:
@harunkovacevic latest stable-backports has now fixes for the D/SHIFTIN1/2 issues.

hansfbaier, 2025-02-24:
The error in this ticket is now fixed, tested with seed 1.
Please open tickets for the remaining issues.

harunkovacevic, 2025-02-25:
Now, I am getting a very similar routing-related error. I am opening a new [issue.](https://github.com/chili-chips-ba/openeye-CamSI/issues/40)

closed 2025-02-25 by harunkovacevic