chili-chips-ba/openeye-CamSI #39: openXC7 - ERROR: Found IDELAYCTRL but no I/ODELAYs in group default

harunkovacevic, 2025-02-08:
After RTL modification related to the _csi_byte_clk_ signal, we managed to bypass the [routing error](https://github.com/chili-chips-ba/openeye-CamSI/issues/38).

> NOTE: For testing purposes only – the error still exists, we just masked it, and it is described in this [issue](https://github.com/chili-chips-ba/openeye-CamSI/issues/38). Modifications are placed in a separate file _**csi_rx_top_2.v**_.

After that, the tool reports a new issue related to IDELAYCTRL component.

`ERROR: Found IDELAYCTRL but no I/ODELAYs in group default
2 warnings, 1 error
make: *** [Makefile:69: error2.fasm] Error 255`

Steps to reproduce:

- Navigate to _3.build/openXC7_, where the Makefile is located.
- Run `make error2`

@hansfbaier, @MikeReznikov Do you have any ideas on how to solve this problem?

hansfbaier, 2025-02-12:
You may circumvent this by commenting out your IDELAYCTRL module. Do you have any I/ODELAYs instantiated?

chili-chips-ba, 2025-02-13:
For this high-speed CSI link that runs at 456MHz DDR (912Mbps/lane), we absolutely need IDELAY component to deskew the 2 data lanes and bring the clock lane right smack in the middle of the eye opening. 

[image]

[image]

- https://github.com/chili-chips-ba/openeye-CamSI/blob/main/0.doc/FPGA-Block-Diagram.png

Barring soldering-down the imaging chip next to FPGA, and careful balancing of PCB traces that connect them, there won't be Hi-Def pictures w/o IDELAY, esp. not in a setup with multiple CRUVI and other connectors in the signal path, in addition to the camera cable.

chili-chips-ba, 2025-02-15:
@regymm any additional tips?

regymm, 2025-02-18:
Again I'm now stuck at the https://github.com/chili-chips-ba/openeye-CamSI/issues/38#issuecomment-2664588954 and can't reproduce this right now. The error text means no IDELAYE2 is instantiated. But I do see the instantiation in csi_rx_phy_dat.v, maybe you can check the nextpnr log to see the device utilisation, like: 
```
Info: 	            IDELAYE2:     9/  500     1%
Info: 	        OLOGICE3_TFF:     0/  500     0%
Info: 	      OLOGICE3_OUTFF:     0/  500     0%
Info: 	       OLOGICE3_MISR:     0/  500     0%
Info: 	           OSERDESE2:    34/  500     6%
Info: 	        ILOGICE3_IFF:     0/  500     0%
Info: 	ILOGICE3_ZHOLD_DELAY:     0/  500     0%
Info: 	           ISERDESE2:    10/  500     2%
Info: 	               BUFIO:     0/   40     0%
Info: 	          IDELAYCTRL:     1/   10    10%
```
to make sure the IDELAYE2 primitives are really used. 

You may have a glance at the UberDDR3 instantiation that (I believe) works with OpenXC7. https://github.com/AngeloJacobo/UberDDR3/blob/main/rtl/ddr3_phy.v#L1062

chili-chips-ba, 2025-02-18:
This design works like a charm with Vivado. 

The whole point of this exercise is not to patch it up to work in openXC7 by avoiding openXC7 toll pitfalls, but to make _openXC7_ better. In that sense, the expected course of action is for the _openXC7_ team to dive into the tool code, looking for the clues about what might be the problem.

chili-chips-ba, 2025-02-18:
Leaving inputs unconnected is a major bug. Something like that is simply not allowed. Vivado will of course tie them off. It'd be interesting to see what Yosys does with them.

It is very surprising that uberDDR3 is leaving HM inputs unconnected. From it, one could say that uberDDR3 project does not run RTL Lint as part of the QA process, and it is not paying attention to Vivado warnings.

chili-chips-ba, 2025-02-18:
@harunkovacevic , @kerimbavcic let's review Yosys logs and file a ticket with them about this undue removal of design elements, the `IDELAYE2` and `ISERDESE2` in particular.

chili-chips-ba, 2025-02-19:
@regymm , @hansfbaier while we are looking at error2, could you please focus on error1?!

hansfbaier, 2025-02-19:
@chili-chips-ba yosys gives me a compile error:
```
$ make error1
...
12. Executing Verilog-2005 frontend: /devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v
Parsing Verilog input from `/devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v' to AST representation.
/devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v:1: ERROR: Re-definition of module `$abstract\csi_rx_top'!
make: *** [Makefile:53: error1.json] Error 1
```

regymm, 2025-02-19:
I'm able to bypass this by renaming 1.hw/csi_rx/csi_rx_top.v to non-verilog file. Then I met the previous-mentioned `ERROR: OSERDESE2 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.slave_oserdese' has disconnected OQ/OFB output ports)`. 

I can change nextpnr code to reduce this to a warning, then I met:
```
Info: Running router2...
Info: Setting up routing resources...
Info:     x splitpoint: 103
Info:     y splitpoint: 91
Info:         bin 0 N=587
Info:         bin 1 N=40
Info:         bin 2 N=16
Info:         bin 3 N=522
Info:         bin 4 N=2447
Info: Running main router loop...
Info: 1186/3612 nets not multi-threadable
ERROR: Failed to route arc 161 of net 'csi_byte_clk', from SITEWIRE/BUFGCTRL_X0Y31/O to SITEWIRE/SLICE_X2Y109/CFFMUX_OUT.
```

Several versions of nextpnr give the same error.

hansfbaier, 2025-02-19:
@harunkovacevic , @regymm This did not work for me. What am I doing wrong here?
```
❯ mv /devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v /devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v.bak

openeye-CamSI/3.build/openXC7 on  main [✘!?] via 🐍 v3.9.12 (venv) via ❄️  impure (shell) 
❯ make
make: *** No rule to make target '/devel/HDL/src/openeye-CamSI/3.build/openXC7/1.hw/csi_rx/csi_rx_top_1.v', needed by 'error1.json'.  Stop.```

regymm, 2025-02-19:
I renamed the csi_rx_top.v instead of the csi_rx_top_1.v.

chili-chips-ba, 2025-02-19:
... this renaming "fix" seems like something that Yosys team should be notified about. @regymm, @hansfbaier do you agree?!

hansfbaier, 2025-02-19:
I thought that was just wrong project setup. The renaming did not work for me. How does it work on your machine @harunkovacevic ?

regymm, 2025-02-19:
This error means yosys found two modules with the same name, and can't
decide which to use. I think it's the default and correct behavior. Vivado
may pick just one according to its own convention.

harunkovacevic, 2025-02-19:
@hansfbaier Original _csi_rx_top.v_ file was unintentionally included in the build. This has now been corrected by explicitly filtering it out in Makefile.

Please try again and let me know if any further issues arise.

chili-chips-ba, 2025-04-03:
@hansfbaier @regymm @edmundhumenberger can we now close this issue?

regymm, 2025-04-03:
In Vivado, optimizing away or ignore IDELAYCTRL when no I/ODELAY is present seems the default behaviour. 
In OpenXC7, this needs to be managed manually now. It has no influence on correctness. Maybe this can be level-ed down from ERROR to a WARNING. 

But I think this can be closed? Later updates can be tracked on https://github.com/openXC7/nextpnr-xilinx/issues/60

chili-chips-ba, 2025-04-03:
Thank you for explanation. With that, we're close our issue and you can continue tracking within openXC7 project.

closed 2025-04-03 by chili-chips-ba