chili-chips-ba/openeye-CamSI #41: openXC7 - ERROR: Failed to route arc 0 of net 'u_hdmi_top.u_hdmi_backend._tdms_sdat[2].u_oser_dat.shift1', from SITEWIRE/OLOGIC_X1Y79/SHIFTOUT1 to SITEWIRE/OLOGIC_X1Y76/SHIFTIN1.

regymm, 2025-03-04:
This happens after https://github.com/chili-chips-ba/openeye-CamSI/issues/40 is work-arounded. The code in question is https://github.com/chili-chips-ba/openeye-CamSI/blob/main/3.build/openXC7/1.hw/lib/fpgatech/XILINX/fpga_oser10.v

The SHIFTIN/SHIFTOUT being connected together seems not supported by OpenXC7. 

As far as I know, this is the first time OSERDESE2 master/slave configuration is used in OpenXC7. 

From https://docs.amd.com/r/en-US/ug953-vivado-7series-libraries/OSERDESE2 : 
```
SHIFTIN1 / SHIFTIN2 	Input 	1 	Cascade Input for data input expansion. Connect to SHIFTOUT1/2 of slave.
SHIFTOUT1 / SHIFTOUT2 	Output 	1 	Cascade out for data input expansion. Connect to SHIFTIN1/2 of master. 
```

hansfbaier, 2025-03-04:
On this one we need to look into whether SITEWIRE/OLOGIC_X1Y79 is placed right with respect to SITEWIRE/OLOGIC_X1Y76.
For that we need to look into the implementation viewer of where the SITEWIRE/OLOGIC_X1Y79/SHIFTOUT1 cables go.
I think they can connect only to one certain module, so the placement of those needs to be done in the packing phase.

chili-chips-ba, 2025-03-04:
Who will be looking into it?!

regymm, 2025-03-19:
I got mixed news! 

Good news is by manually placing the OSERDESE2 instances at the correct location, this shiftin/shiftout problem is solved. 
This can be done by creating a constraints.py with the following content:
```python
# !/usr/bin/env python3
# Original code from Hans Baier
def get_cells(name_part):
    return list(map(lambda c: c.second, filter(lambda c: name_part in c.first, ctx.cells)))

def get_cell(name_part):
    return get_cells(name_part)[0]

c2m_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[2].u_oser_dat.master_oserdese'
c2s_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[2].u_oser_dat.slave_oserdese'
c2m=get_cell(c2m_name)
c2s=get_cell(c2s_name)
c1m_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[1].u_oser_dat.master_oserdese'
c1s_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[1].u_oser_dat.slave_oserdese'
c1m=get_cell(c1m_name)
c1s=get_cell(c1s_name)
c0m_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.master_oserdese'
c0s_name = 'u_hdmi_top.u_hdmi_backend._tdms_sdat[0].u_oser_dat.slave_oserdese'
c0m=get_cell(c0m_name)
c0s=get_cell(c0s_name)

c0m.setAttr('BEL', 'OLOGIC_X1Y80/OSERDESE2')
c0s.setAttr('BEL', 'OLOGIC_X1Y79/OSERDESE2')
c1m.setAttr('BEL', 'OLOGIC_X1Y94/OSERDESE2')
c1s.setAttr('BEL', 'OLOGIC_X1Y93/OSERDESE2')
c2m.setAttr('BEL', 'OLOGIC_X1Y76/OSERDESE2')
c2s.setAttr('BEL', 'OLOGIC_X1Y75/OSERDESE2')

```
And slightly change the nextpnr line in Makefile:
```
	/mnt/PTP/zynq-openxc7/nextpnr-xilinx/build/nextpnr-xilinx --chipdb ${CHIPDB}/${PART}.bin --xdc $(XDC_FILE) --json error1.json --pre-place constraints.py --fasm $@ --verbose #--debug

```
So the OSERDESE2s are manually placed. The correct value can be found in Vivado open implemented design, go to the pins the oserdese2 is connected to, and see the (small) logic and its location. Be aware that the [0], [1], [2] in the generated module is somehow inverted with the HDMI pinout index [2], [1], [0]. 

[image]

regymm, 2025-03-19:
The bad news is bitstream can't be generated yet due to missing entries in prjxray-db. 
I got this log for nextpnr-xilinx and fasm:
```
Info: Running post-routing legalisation...
Warning: Unprocessed route-thru LIOI3_X0Y109.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1
!Warning: Unprocessed route-thru LIOI3_X0Y117.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1
!4 warnings, 0 errors
fasm2frames --part xc7a100tcsg324-2 --db-root /mnt/PTP/zynq-openxc7/nextpnr-xilinx/xilinx/external/prjxray-db/artix7 error1.fasm > error1.frames
/prjxray/third_party/fasm/fasm/parser/__init__.py:30: RuntimeWarning: Unable to import fast Antlr4 parser implementation.
  ImportError: cannot import name 'antlr_to_tuple' from partially initialized module 'fasm.parser' (most likely due to a circular import) (/prjxray/third_party/fasm/fasm/parser/__init__.py)

  Falling back to the much slower pure Python textX based parser
  implementation.

  Getting the faster antlr parser can normally be done by installing the
  required dependencies and then reinstalling the fasm package with:
    pip uninstall
    pip install -v fasm

  warn(
Traceback (most recent call last):
  File "/prjxray/env/bin/fasm2frames", line 33, in <module>
    sys.exit(load_entry_point('prjxray', 'console_scripts', 'fasm2frames')())
  File "/prjxray/utils/fasm2frames.py", line 304, in main
    run(
  File "/prjxray/utils/fasm2frames.py", line 189, in run
    assembler.parse_fasm_filename(filename_in, extra_features=extra_features)
  File "/prjxray/prjxray/fasm_assembler.py", line 203, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB RIOI3, key RIOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y79.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3, key RIOI3.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y79.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3_TBYTESRC, key RIOI3_TBYTESRC.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_TBYTESRC_X57Y93.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3_TBYTESRC, key RIOI3_TBYTESRC.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_TBYTESRC_X57Y93.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3, key RIOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y75.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3, key RIOI3.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y75.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3, key RIOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y65.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB RIOI3, key RIOI3.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y65.OLOGIC_Y1.OSERDES.DATA_WIDTH.DDR.W10'
Segment DB LIOI3, key LIOI3.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1 not found from line 'LIOI3_X0Y109.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1'
Segment DB LIOI3, key LIOI3.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1 not found from line 'LIOI3_X0Y117.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1'
make: *** [Makefile:59: error1.frames] Error 1
make: Leaving directory '/mnt/SymbioticEDA/openeye-CamSI/3.build/openXC7'
exit status 2
```

- Firstly, I've never seen anything like `Warning: Unprocessed route-thru LIOI3_X0Y109.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1`. 
This seems closely related to the last parts of fasm error
```
Segment DB LIOI3, key LIOI3.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1 not found from line 'LIOI3_X0Y109.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1'
Segment DB LIOI3, key LIOI3.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1 not found from line 'LIOI3_X0Y117.LIOI_OLOGIC0_OFB.IOI_OLOGIC0_D1'
```

- Then, many W10 related entries are missing. According to official oserdese2 document Available Attributes, DATA_WIDTH https://docs.amd.com/r/en-US/ug953-vivado-7series-libraries/OSERDESE2, "When DATA_RATE_OQ is DDR, the possible values for the DATA_WIDTH attribute are 4, 6, 8, 10 and 14." 
Fasm also emits W10, however, W10 and W14 entries are not in prjxray-db. 

Wondering if @hansfbaier has any ideas on this?

hansfbaier, 2025-03-19:
> c0m.setAttr('BEL', 'OLOGIC_X1Y80/OSERDESE2')
> c0s.setAttr('BEL', 'OLOGIC_X1Y79/OSERDESE2')
> c1m.setAttr('BEL', 'OLOGIC_X1Y94/OSERDESE2')
> c1s.setAttr('BEL', 'OLOGIC_X1Y93/OSERDESE2')
> c2m.setAttr('BEL', 'OLOGIC_X1Y76/OSERDESE2')
> c2s.setAttr('BEL', 'OLOGIC_X1Y75/OSERDESE2')

This can be automated like I have been done in 
pack_dsp_xc7.cc:162 and 66.
Reason is that the SHIFTIN/OUT lines only run to the site
directly above/below it.

hansfbaier, 2025-03-19:
> prjxray.fasm_assembler.FasmLookupError: Segment DB RIOI3, key RIOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 not found from line 'RIOI3_X57Y79.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10'
> Segment DB RIOI3, key [...]
[...]

One hack you can try is to blacklist the pip in arch.cc:342.
That means the router won't consider it and find another way around that pip.
But that also can result in no route being found, if there are no alternative paths
and routing will fail.
The proper way, of course is to make the fuzzer find the pip, which
I am also currently struggling with at GTX.

chili-chips-ba, 2025-03-20:
> "... the proper way, of course is to make the fuzzer find the pip..."

@regymm do you know how to run the fuzzer to try this?!

regymm, 2025-03-20:
I saw the [fuzzers/036-iob-ologic/generate.py](https://github.com/openXC7/prjxray/blob/master/fuzzers/036-iob-ologic/generate.py#L45) fuzzer contains the DATA_WIDTH settings. However, after fuzzing, this segment
```
IOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W10 <const0>
IOI3.OLOGIC_Y0.OSERDES.DATA_WIDTH.DDR.W14 <const0>
```
is found in fuzzers/036-iob-ologic/build/segbits_xioi3.rdb, and no related entries are ended in the final database. 

Does this mean the W10 entries can be safely ignored?

chili-chips-ba, 2025-04-03:
@hansfbaier @edmundhumenberger @regymm it seems that we are very close, and only one push away. Is there anything we can do to help you tilt this boulder off the road that it's been blocking?!

hansfbaier, 2025-05-28:
@regymm, @chili-chips-ba  the 10 and 14 data widths are only for cascaded SERDES. They are not at all exercised by the fuzzer, if you look into top.py. It is very likely that those will be a combination of the lower DDR.W* but for master and slave SERDES combined.
If you can build me a minimal example with one master and one slave SERDES and DATA_WIDTH 10 in vivado,
then I can disassemble that and see what is done, and support it in nextpnr-xilinx

[image]

chili-chips-ba, 2025-05-28:
Hi @hansfbaier, please use [fpga_oser10.sv](https://github.com/chili-chips-ba/openeye-CamSI/blob/main/1.hw/lib/fpgatech/XILINX/fpga_oser10.sv)

hansfbaier, 2025-05-29:
@regymm There are actually bits missing for MASTER/SLAVE setup with bitwidths 10 and 14:
Left is 10 and right is 14 bitwidth. Those disappear, when no MASTER/SLAVE setup is used.

[image]

chili-chips-ba, 2025-06-26:
Closing, since now fixed and merged in prjxray by @hansfbaier. For more, see https://github.com/f4pga/prjxray/pull/2550

closed 2025-06-26 by chili-chips-ba